Jinjinov/Usb.Events #1: Not all USB based TTY (serial) devices enumerate or fire arrival/removal events

maskimthedog, 2021-01-28:
I have a C# application that utilizes udev to enumerate and receive events from USB based TTY (serial) devices and receive arrival/removal events.  Your solution covers Windows/Mac/Linux and it is well implemented.  However, it does not enumerate nor fire events for all the USB based TTY devices I use on my Linux system, devices with ttyACM* and ttyUSB*.  I found that I have to filter on "tty" subsystem to enumerate/receive events for these devices.

Jinjinov, 2021-01-28:
I don't have any TTY device, so I can't test my code if it works correctly for TTY.

I don't know anything about TTY devices, so if you can give me any good links about working with TTY in Windows, Mac, Linux and if you can suggest an interesting and fun TTY device I could buy for less than 50 euros (so that I can test any detection code I write), I will try to add TTY detection to my library :)

maskimthedog, 2021-01-29:
Checkout my fork at https://github.com/maskimthedog/Usb.Events.  I am new to C# so it needs refactoring most likely.

Also, I will send contact you OOB and purchase two different TTY devices for you and have them delivered directly to you.

closed 2021-02-02 by Jinjinov