Jinjinov/Usb.Events #2: How are you building and packing

KZeronimo, 2021-01-29:
Hi @Jinjinov  - nice job on the library!

Are you building on each native system, copying the resulting binary and then packing on window?

I'm thinking of the flow as

dotnet build on linux and output binary as artifact
dotnet build on mac and output binary as artifact
copy binary artifacts from each agent to windows agent
dotnet pack on windows

More curious than anything - thanks in advance!

Jinjinov, 2021-01-29:
Thank you, I am glad you like it!

Yes, that is exactly what I am doing. I write the code in Windows and I share the solution folder with a Linux virtual machine, so that when I build in Linux, the built binary is already in the correct folder.

KZeronimo, 2021-01-29:
Awesome - thanks for the quick response - on the topic of building and artifacts have you considered using semver for the version number major.minor.patch - seems more canonical in the .NET shared lib world

Jinjinov, 2021-01-29:
I have, in fact, considered it :)

I admit that according to https://semver.org/ my versioning violates one rule of the Backus–Naur Form Grammar for Valid SemVer Versions. That one rule is that the last separator should be a - or a + but not a dot.

On the other hand, in the Visual Studio IDE in the Package tab of the project Properties window, there are four text input boxes for the Assembly version. When you use them, VS writes the four values separated by dots, so in the end I decided to use that.

I agree that it is not canonical, but I find the binary version notation amusing :) My little joke in the otherwise depressing 2020.

closed 2021-02-02 by Jinjinov