Jinjinov/Usb.Events #8: Add capability to specify subsystem for Linux udev.

maskimthedog, 2021-01-31:
ttyACM and ttyUSB reside under the "tty" subsystem on Linux and are not currently discoverable via Usb.Events since it sets the subsystem to "usb".

Jinjinov, 2021-01-31:
Thank you for all your contributions, I really appreciate it!

I have made some commits to my repository for the other issues. Please close them if they are resolved, or comment what still needs to be done.

I see that you made a new "string subsystem" parameter for the "UsbEventWatcher" class, but that means that it can report events for either "usb" or "tty", but not both - unless specifying "tty" also includes all "usb" events? I don't know if that is a desired behavior for other users of this library. Wouldn't it be better if both types were reported simultaneously? I also don't yet know how to implement this in Windows and MacOS, I will have to take a look. I will also buy a TTY device so I can test any code I write for Windows and MasOS. I will take a look at what devices are available today.

maskimthedog, 2021-01-31:
No problem.  Yes, having a special case for Linux breaks the interface.  I have never tried filtering on to susbsystem at the same time for enumerating or hotplugging on UDev.  That would be one option.  Coolgear has a lot of different USB to serial cables.  I have no problem giving you some Euros to purchase a couple.  :)

https://www.coolgear.com/product-category/adapters/usb?gclid=Cj0KCQiAx9mABhD0ARIsAEfpavReMZxOLXRdrJ-yj8uVHV9Aesnl_zUbOjTYgwtMIpND29x6XESnPs0aAp7REALw_wcB

maskimthedog, 2021-01-31:
I just inserted a subsystem of "tty" along with a subsystem of "usb" in UsbEventWatcher.Linux.c.  It appears to work fine to have UDev filtering on multiple subsystems for both the enumeration and monitor methods.  Doing this and not passing down a settable subsystem from the C# code does not pollute the interface for the other two supported OSes by adding a special case which they would not care about.

Jinjinov, 2021-02-01:
I asked my coworker from the hardware department if I could borrow a USB to serial cable and he said that I could. I didn't even know they had these cables at my workplace :) Thank you for your offer anyway!

That is great to hear! Make a pull request when you are ready. Since I have already merged your changes for the other issues, there should not be so many changed lines this time.

Jinjinov, 2021-02-01:
Can you test the new NuGet package?

closed 2021-02-02 by Jinjinov