Jinjinov/Usb.Events #12: Not fired add but it works when remove

GoNzCiD, 2021-05-14:
**Library version**
1.1.1.1

**OS & OS version**
Windows 10 Pro (19042.985)

**Describe the bug**
I connect/disconect some USB devices, and it only registers events when I remove it but not when I connect it to the PC.

**To Reproduce**
Steps to reproduce the behavior:
I'm just importing the library via nuget and running the sample source code from the readme.md. And then plug/unplug some USB devices.

**Expected behavior**
See DeviceAdded events

Jinjinov, 2021-05-14:
Can you give me any more details about the devices that are not recognized? Device type, model, manufacturer - or device specifications website?

If I can't reproduce the bug, I can't fix it :)

GoNzCiD, 2021-05-14:
Yes, please tell me if this info is enough:
* A Toshiba memory stick 16Gb (Regarding google model seems to be hayabusa):
  This is the lines when I unplug it, but no logging when I plug-in it:
  ```
  Removed:
  Device Name:
  Device System Path: USB\VID_0E0F&PID_0001\66259FFDB0A0CD9107187694
  Mounted Directory Path:
  Product:
  Product Description:
  Product ID: 0001
  Serial Number: 66259FFDB0A0CD9107187694
  Vendor:
  Vendor Description:
  Vendor ID: 0E0F
  ```
And the same for:
* [this](https://aliexpress.com/item/1005001571293451.html) USB KVM
* Logitec Unifying USB receiver: https://www.logitech.com/products/mice/unifying-receiver-usb.910-005236.html
* Arduino UNO (Elegoo R3) https://www.elegoo.com/products/elegoo-uno-r3-board?_pos=1&_sid=6debb4cda&_ss=r
* Generic (Unknown manufacturer and model) smart card reader

And the same for all, it can recognize when I remove the device but not when I plug them in :(

Jinjinov, 2021-05-14:
I see, thanks for the details. Unfortunately I don't own any of the devices you listed, except for a Toshiba memory stick which works on my PC. Are any USB devices detected at all? If not, there could be a problem with Windows 10 (I use Windows 8.1)

### Does any of the previous NuGet packages work?

I changed the detection code a few times, so previous code could actually work.

### If none of the previous NuGet packages work, you can try uncommenting these lines:

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L232
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L240

and comment these lines:

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L234
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L242

The program will probably crash, but you should at least get the event, if you put a breakpoint at

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L297

GoNzCiD, 2021-05-14:
Hi @Jinjinov,

### I don't test with older versions
This is the first time I use your package, I'm just looking for an event that alerts me when the Arduino is connected to PC1 or PC2 via the KVM USB of the link above. (In fact it should be valid for me the USB hub detection, not needed the end USB device)
**I can try with older versions in the next days, but I have discovered an interesting behavior...**

**Scenario**: The KVM USB connected to one USB PC PORT and some devices connected to the KVM USB (even a USB HUB to the KVM and other end device to this HUB). I'm running the last sources from this repo.

**Test 1**:
* Plug the KVM to the PC.
* No events detected
* Plug the memory stick (or any other end device) to other USB PC port
* All `Added` events are shown, from the memory stick and any other devices connected (KVM, HUB, and end devices)
* Unplug the KVM
* All the `Removed` events are shown

**Test 2**:
1) The KVM is plugged
1) Swap the KVM to the other PC
1) All the `Removed` events are shown
1) Swap the KVM to the current PC
1) No `Added` events detected but I can see in log console an entry for an USB Concentrator.
1) Swap the KVM to the other PC again
1) All the `Removed` events are shown
1) Swap the KVM to the current PC again
1) No events nor log console events are shown
1) I can loop from 6 to 9 with the same results
1) Plug an end device to one USB PC port
1) All Added events are shown (a lot of them, probably repeated, it seems like they are queued in some buffer and once I use another USB port of the PC all of them are processed at this moment....)
1) And now I'm at point 1 again.

I don't know if it helps you in some way or do you have any other test in mind that I can do...

Jinjinov, 2021-05-14:
Oh, so, you are using your USB devices via the KVM switch - that explains a lot.

My `Usb.Events` works with `WMI` in Windows and I think that `WMI` triggers one event for one `Win32_USBControllerDevice`

Because you have your KVM plugged in, there is only one such event, because there is only one `Win32_USBController` - all other devices are child devices and the KVM is the parent device.

`Usb.Events 1.1.1.1` listens to `Win32_USBControllerDevice` events, but some older versions listened to the `Win32_PnPEntity` events, so they might work.

Try with older NuGet versions - or try to comment the `Win32_USBControllerDevice` lines and uncomment the `Win32_PnPEntity` like I said in the previous comment.

closed 2021-05-24 by Jinjinov

GoNzCiD, 2021-06-01:
Unfortunately I have not enough time to work on it, but I could test that changing these lines and debug, some even is catch when plug/unplug devices to the kvm, So maybe yes, it's possible to adapt it.