Jinjinov/Usb.Events #14: Disposing UsbEventWatcher on Linux throws a null reference exception.

thomOrbelius, 2021-07-01:
**Library version**
1.1.1.1

**OS & OS version**
Ubuntu Server 20.04 LTS

**Describe the bug**
When disposing UsbEventWatcher on Linux we get a null reference exception. 

`System.NullReferenceException : Object reference not set to an instance of an object.
           at Usb.Events.UsbEventWatcher.Dispose()
`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a disposable class with the UsbEventWatcher in it and dispose it (see code below)

(We used Dependency Injection when we found it, but it's easily reproducible by just calling the dispose method)

    internal class Dummy : IDisposable
    {
        private readonly UsbEventWatcher _usbEventWatcher = new();
        private bool _disposedValue;

        protected virtual void Dispose(bool disposing)
        {
            if (!_disposedValue)
            {
                if (disposing)
                {
                    _usbEventWatcher.Dispose();
                }

                _disposedValue = true;
            }
        }
    }

**Expected behavior**
The Dispose method should not throw any null reference exceptions.

**Additional context**
I think it's the Windows specific objects `_volumeChangeEventWatcher, _USBControllerDeviceCreationEventWatcher, _USBControllerDeviceDeletionEventWatcher` that throws the exception as they are not created on the other platforms. 
A simple null check per object or a Windows specific dispose section should fix it. 

I do have one concern about this - are there any Linux/Mac specific objects that also needs to be disposed here?

Jinjinov, 2021-07-01:
Thank you, I will look into this as soon as I have time, hopefully in a few days.

Regarding the disposal of any Linux/Mac specific objects

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.Linux.c
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.Mac.c

The code for Linux and Mac is pure C, because it is not possible to do this in C#

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L34

as you can see here, the project file compiles the C code to .so on Linux and to .dylib on Mac

As I am not a C programmer, I wrote the C code with lots and lots of Stackoverflow help - I would say that there is no need to dispose of any C objects, but I am not 100% certain.

thomOrbelius, 2021-07-02:
I made a simple fix, where the objects are null checked before disposing.

I have not been able to test in for Mac, but the updated version of the test application should be easy to use.

closed 2021-07-02 by Jinjinov