Jinjinov/Usb.Events #17: Inside the windows service this package not working

spark6-dev, 2021-09-07:
### Inside the windows service it does not trigger any events with this package.
When we tested this package, it worked in the console application. But this does not work within the Windows services. Is there are any restrictions for this to work inside the Windows services?

Jinjinov, 2021-09-07:
> Events then initialize the functionality inside the OnStart event

These are not exact steps to reproduce the problem.

Please provide exact code to reproduce the problem.

spark6-dev, 2021-09-08:
### To Reproduce

-     In Visual Studio 2019, we created a project for Windows services
-     We added the following code blocks to the Windows service's OnStart method
-     `try
{
	EventLog.WriteEvent("TestService", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { "Service start successfully." });
	using (IUsbEventWatcher usbEventWatcher = new UsbEventWatcher())
	{
		EventLog.WriteEvent("TestService", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { usbEventWatcher.UsbDeviceList.FirstOrDefault().DeviceName.ToString() });
		Console.WriteLine(usbEventWatcher.UsbDeviceList.FirstOrDefault());
		usbEventWatcher.UsbDeviceRemoved += (_, device) => EventLog.WriteEvent("TestService", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { "Removed:" + Environment.NewLine + device + Environment.NewLine });

		usbEventWatcher.UsbDeviceAdded += (_, device) => EventLog.WriteEvent("TestService", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { "Added:" + Environment.NewLine + device + Environment.NewLine });

		usbEventWatcher.UsbDriveEjected += (_, path) => EventLog.WriteEvent("TestService", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { "Ejected:" + Environment.NewLine + path + Environment.NewLine });

		usbEventWatcher.UsbDriveMounted += (_, path) =>
		{
			Console.WriteLine("Mounted:" + Environment.NewLine + path + Environment.NewLine);
			EventLog.WriteEvent("Mounted", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { "Ejected:" + Environment.NewLine + path + Environment.NewLine });
			foreach (string entry in Directory.GetFileSystemEntries(path))
				EventLog.WriteEvent("TestService", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { "Entries:" + Environment.NewLine + entry + Environment.NewLine });
		
		};
	}
}
catch (Exception ex)
{
	EventLog.WriteEvent("TestService", new EventInstance(0, 0, EventLogEntryType.Information), new string[] { ex.Message });
}`
-     However, the events were not triggered. The function was tracked using the event logger, but there are no logs in the event viewer.
-     We missed anything here?

Jinjinov, 2021-09-08:
`IUsbEventWatcher usbEventWatcher` starts listening for USB events from the moment you call `usbEventWatcher = new UsbEventWatcher()` and it stops listening for USB events when you call `usbEventWatcher.Dispose()`

`using (IUsbEventWatcher usbEventWatcher = new UsbEventWatcher())` calls `usbEventWatcher.Dispose()` when the execution reaches `}` because that is how `using` works:

https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/keywords/using-statement

solution:
- make `IUsbEventWatcher usbEventWatcher` a  private member of the Windows service class
- call `usbEventWatcher = new UsbEventWatcher()` inside `OnStart` - without `using`
- call `usbEventWatcher.Dispose()` inside `OnStop`

spark6-dev, 2021-09-08:
@Jinjinov: **Your efforts are greatly appreciated and thank you for providing some valuable solutions on time. It worked great and your quick response was really appreciated.**

closed 2021-09-08 by spark6-dev