Jinjinov/Usb.Events #18: USB exact description need to include.

spark6-dev, 2021-09-08:
**Actually most of the device doesn't have any product name or vendor name. but still, vendors added some information inside the "Bus reported device description"**

Some of the devices only included the vendor and product information, but in most of the cases these values will be default ones like the product: "USB Input Device",  vendor: "Standard System device".

**Solution recommended**

If possible we can include the "Bus reported device description" into the below query.

```
 using ManagementObjectSearcher Win32_PnPEntity = new ManagementObjectSearcher($"SELECT Caption, ClassGuid, Description, DeviceID, Manufacturer FROM Win32_PnPEntity WHERE DeviceID LIKE '%{serial}%'");

```

This is the right method to identify the name for the devices most of the other keys are default ones so we are unable to find the exact devices.

This "Bus reported device description" can be set by the USB device, for instance

Jinjinov, 2021-09-08:
I searched for a solution, but I only found one possible answer that could be used with WMI:

https://stackoverflow.com/a/59282440/4675770

As you can see, others are using native Windows libraries with `[DllImport]`:

https://stackoverflow.com/a/36616242/4675770

and that method is not compatible with WMI which is used in Usb.Events

I will try to find a way to make it work with WMI, but I won't include `[DllImport]` into Usb.Events.

Jinjinov, 2021-09-08:
It seems like we need https://docs.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdigetdevicepropertyw

and https://docs.microsoft.com/en-us/windows-hardware/drivers/install/devpkey-device-busreporteddevicedesc

but the problem is, how to call the C++ `SetupDiGetDeviceProperty` with the data we get from C# `ManagementObjectSearcher`

Jinjinov, 2021-09-08:
There are two main problems here:
- Not all USB devices have the same information stored in WMI (many are missing certain data)
- I don't have the USB devices that you are using

When you query `Win32_PnPEntity` you can get more than one result containing `Caption`, `Description`, `Manufacturer`

I have chosen the best probable result by trial and error, as you can see here:

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L419

You can comment out this line:

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L423

And put a breakpoint here:

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/UsbEventWatcher.cs#L408

This breakpoint can be hit more than one time in some cases. You can check if there is a better description in any other cases.

Jinjinov, 2021-09-08:
I won't have access to my PC for the next few days.

closed 2021-09-10 by spark6-dev

Jinjinov, 2021-09-11:
@spark6-dev 

I found a solution: https://github.com/Jinjinov/Usb.Events/tree/master/Usb.Events.Example