Jinjinov/Usb.Events #19: false "device added" events are fired on Linux

d79ima, 2021-10-29:
### Library version
10.0.0

### OS & OS version
Ubuntu 18.04 LTS

### Describe the bug
When i unplug my USB device from Linux machine i am seeing Usb.Events fire an UsbDeviceAdded event.
I monitored udev events on the Linux machine using "udevadm monitor -e" and I can clearly see that no "add" events are generated. I see "remove" and "unbind" events only.

In your code in UsbEventWatcher.Linux.c line 141, i assume that if event isn't a "remove" then it must be an "add", which as you can see is a bug.
Could you please fix this and publish and updated Nuget please, basically just need to put in a check like this:
(strcmp(action, "remove") == 0 || strcmp(action, "unbind") == 0)

Otherwise, great library and thanks for writing it.

### To Reproduce
reproducing this should be easy at least on Ubuntu 18.04. Not sure about other distros...

Jinjinov, 2021-10-30:
Thank you for noticing this.

When I initially researched the `udev_device_get_action` the example I found only mentioned `add` and `remove` actions.

I just researched it again and found this:

https://github.com/systemd/systemd/blob/main/src/libudev/libudev-device.c

> Usual actions are: add, remove, change, move, online, offline.

I agree with you that the condition check must be updated, but the `unbind` action is not mentioned in there.

I only found one mention of the `unbind` action: in an issue that describes breaking the event handling with kernel 4.12

https://github.com/systemd/systemd/issues/8221

It seems like `bind` and `unbind` were added in kernel 4.12

Do you have any suggestions on how this should be handled? I don't have a lot of free time for research right now.

Do you think it should be fixed like this:

    if (action && strcmp(action, "remove") == 0)
    {
        RemovedCallback(usbDevice);
    }
    else if (action && strcmp(action, "add") == 0)
    {
        InsertedCallback(usbDevice);
    }

or like this:

    if (action && (strcmp(action, "remove") == 0 || strcmp(action, "unbind") == 0 || strcmp(action, "offline") == 0))
    {
        RemovedCallback(usbDevice);
    }
    else if (action && (strcmp(action, "add") == 0 || strcmp(action, "bind") == 0 || strcmp(action, "online") == 0))
    {
        InsertedCallback(usbDevice);
    }

d79ima, 2021-10-30:
I would really love it if you did your second proposal.
`if (action && (strcmp(action, "remove") == 0 || strcmp(action, "unbind") == 0 || strcmp(action, "offline") == 0))
{
    RemovedCallback(usbDevice);
}
else if (action && (strcmp(action, "add") == 0 || strcmp(action, "bind") == 0 || strcmp(action, "online") == 0))
{
    InsertedCallback(usbDevice);
}`

Because for sure at least on the platform i am using "bind" and "unbind" happen consistently and sometimes i only get them instead of "add" or "remove". 
Also could you please share your Makefile for building UsbEventWatcher.Linux.c. This will come in handy in case i need to make changes myself in the future.

Jinjinov, 2021-10-30:
Ok, I will fix this when I get the time, perhaps tomorrow, perhaps next week.

Unfortunately I don't have a Makefile for `UsbEventWatcher.Linux.c` but you can see how I make `UsbEventWatcher.Linux.so` in line 40 of my project file:

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L40

Jinjinov, 2021-10-31:
Can you confirm that this issue is fixed in the new NuGet package?

d79ima, 2021-11-01:
Just checked latest nuget. The removed/inserted device monitoring is now working better.
How ever, i believe the change you made now broke EnumerateDevices. Previously i used to see some devices that i need to know about that are static coming into my app through the Inserted event because you just had a flat "else".
But now you have a "else if". 
WOuld it be possible to change it so that enumerate devices just calls Inserted event unconditionally. So kind of separate logic from monitoring devices coming and going.

Jinjinov, 2021-11-01:
I suspected that there was a reason for `else` without `if` as I even wrote a comment, but is seems it was not detailed enough :)

I will fix this when I have the time.

d79ima, 2021-11-01:
I made a quick fix and tested it. In case you want to use it
[UsbEventWatcher.Linux.c.zip](https://github.com/Jinjinov/Usb.Events/files/7455188/UsbEventWatcher.Linux.c.zip)
.

Jinjinov, 2021-11-01:
I did some additional cleanup since I was already refactoring.

I hope it works now.

d79ima, 2021-11-01:
Just tested it. it works great! thanks very much.

Jinjinov, 2021-11-01:
You're welcome. Thank you for using my library!

closed 2021-11-01 by Jinjinov