Jinjinov/Usb.Events #20: Separate UsbEventWatcher construction from start of enumerating devices

d79ima, 2021-11-09:
### Library version
10.0.1

### OS & OS version
Ubuntu 18.04 LTS

### Describe the bug

### Suggestion

Jinjinov, 2021-11-09:
Thank you for noticing this.

I agree that constructor shouldn't call Start() before you have the chance to subscribe to events.

Unfortunately removing Start() from constructor is a breaking change for all who use my library - if they don't call Start() it will just stop working.

I can put another `Action callback` parameter in constructor, to let you subscribe to events in constructor.
Or perhaps I will add `bool startWatcher = true` to constructor and it will work for all other users, and you can set it to false.

About separating EnumerateDevices from StartMonitoring - I agree that your suggestions would make the library better :)
I would also have to do that for Windows and for MacOS - the library has to be consistent - but unfortunately I don't have the time to do that now.

d79ima, 2021-11-09:
Thank you, i understand. Let's go with` bool startWatcher = true ` for now as a constructor arg. This will be a pretty simple change

Jinjinov, 2021-11-09:
Ok, thanks :) I published a new NuGet .

closed 2021-11-09 by Jinjinov