Jinjinov/Usb.Events #21: Capture device number and bus number on Linux

d79ima, 2022-03-23:
**Is your feature request related to a problem? Please describe.**
I would like to have access to the Linux system attributes Device Num and Bus Num for usb devices.

**Describe the solution you'd like**
Would you be able to add calls in UsbEventWatcher.Linux.c GetDeviceInfo():
udev_device_get_sysattr_value(dev, "devnum")
udev_device_get_sysattr_value(dev, "busnum")

And add 2 new fields into the common UsbDevice class. I am not sure if these fields make sense on Windows and Mac also, they may not be available. In which case can just put a disclaimer in the comments

Jinjinov, 2022-03-23:
You can get the Device Number and Bus Number from the Device Name:

`Device Name: /dev/bus/usb/001/003`

Bus Number is 1
Device Number is 3

```
string[] split = DeviceName.Split('/', System.StringSplitOptions.RemoveEmptyEntries);
string busNum = split[3];
string devNum = split[4];
```

d79ima, 2022-03-23:
perfect thanks

closed 2022-03-23 by Jinjinov