Jinjinov/Usb.Events #23: No events or device list on macOS

hunsra, 2022-06-04:
### Library version
10.0.1.1

### OS & OS version
macOS Monterey Version 12.4

### Describe the bug

No events are invoked on macOS when USB devices are added or removed, and the list of USB devices is always empty.

### To Reproduce

You can reproduce the issue with the following public repo on GitHub:
[https://github.com/hunsra/UsbEventsTest](https://github.com/hunsra/UsbEventsTest)

Steps to reproduce the behavior:
1. Clone the repository above.
2. Open the repository with Visual Studio for Mac version 17.0.1 (build 72).
3. Select the Debug configuration and build the solution.  Note that you will see in the build output the following warning, which may be a clue to what is happening:
```
MMP : warning MM2006: Native library 'UsbEventWatcher.Linux.so' was referenced but could not be found.
MMP : warning MM2006: Native library 'UsbEventWatcher.Mac.dylib' was referenced but could not be found.
    2 Warning(s)
    0 Error(s)
```
4. Run the program and attach a USB device to the Mac computer you are running the program on.

### Exceptions (if any)

### Expected behavior

Based on the description of the library, I should see a list of attached devices in the debugger output window when the program starts, and I should see debugger output when USB devices are added or removed.

### Additional context

My suspicion is that the mac and linux native libraries are not included in the NuGet package, or there is a path issue in the NuGet package configuration.

Jinjinov, 2022-06-04:
The native libraries `UsbEventWatcher.Linux.so` and `UsbEventWatcher.Mac.dylib` are in the NuGet package.

You can check it yourself with https://github.com/NuGetPackageExplorer/NuGetPackageExplorer

[image]

I never tried building with `Visual Studio for Mac`

How it works when you build with `Visual Studio Code`:

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L37
This line uses `gcc` to build `UsbEventWatcher.Mac.dylib` from `UsbEventWatcher.Mac.c` when run on macOS.

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L40
This line uses `gcc` to build `UsbEventWatcher.Linux.so` from `UsbEventWatcher.Linux.c` when run on Linux.

hunsra, 2022-06-04:
@Jinjinov 
Thanks for the follow up.  I think in Visual Studio for Mac, you need to add a "Native Reference" to the dylib in the native folder:
[image]
It works after doing that.  I suspect you'd need to do the same for Linux with anything other than VS code.

closed 2022-06-04 by hunsra