Jinjinov/Usb.Events #26: Can we allow the monitor loop to exit?

zheng-yu-yang, 2023-05-18:
Hi there,

In function MonitorDevices there is a loop to 'select' on the udev fd, which works well but there is no way to exit from this loop. So once StartLinuxWatcher is called we can never stop the wrapping task created by UsbEventWatcher.Start. This is not so good if we want to the monitoring to stop at some point of the program. It also forces the instance of UsbEventWatcher to be a singleton otherwise there might be multiple loops active at the same time and report USB events repeatedly.

I'm wondering, maybe we can add a returning value to UsbDeviceCallback (which returns void by now), and exit from the loop upon a specific value.

Is this a good idea? If yes I can riase a PR.

Regards,

Jinjinov, 2023-05-18:
Yes, it is true that it is not possible to stop monitoring for USB events. It would be possible to implement this, but it would require a lot of work. I want that `Usb.Events` behaves the same in Windows, Linux and macOS. You can just call `Dispose` on `IUsbEventWatcher` bacause you have `interface IUsbEventWatcher : IDisposable` or you can un-subscribe to events to ignore them:

        event EventHandler<string>? UsbDriveMounted;
        event EventHandler<string>? UsbDriveEjected;

        event EventHandler<UsbDevice>? UsbDeviceAdded;
        event EventHandler<UsbDevice>? UsbDeviceRemoved;
        
There should always be only one instance of IUsbEventWatcher - it already reports all USB events - there is no need to have multiple instances. If you would make multiple instances, you would probably break something, or at least, have very strange behavior. In Windows, you would have multiple `EventArrivedEventHandler` for the same event, in macOS you would open multiple `IONotificationPortRef` and in Linux you would create multiple `udev*` - none of these are good ideas.

zheng-yu-yang, 2023-05-18:
I don't think disposing a UsbEventWatcher is a good choice because it leaves the task on StartLinuxWatcher, making creating a new instance of UsbEventWatcher problematic. As to unsubscribing the events, it does not introduce duplicated reports, but the task on GetLinuxMountPoint still runs every 1000ms even I don't need it at all.

Jinjinov, 2023-05-18:
Would it be OK, if I changed 

    Task.Run(() => StartLinuxWatcher(InsertedCallback, RemovedCallback, includeTTY));

to 

    Task.Run(() => StartLinuxWatcher(InsertedCallback, RemovedCallback, includeTTY), _cancellationTokenSource.Token);

This would be much easier than changing the native C code in Linux.

This should work because `IUsbEventWatcher` has:

        public void Dispose()
        {
            _cancellationTokenSource.Cancel();
            _cancellationTokenSource.Dispose();
        }

Jinjinov, 2023-05-18:
I will make a native C solution for Linux and macOS.

zheng-yu-yang, 2023-05-19:
> Would it be OK, if I changed
> 
> ```
> Task.Run(() => StartLinuxWatcher(InsertedCallback, RemovedCallback, includeTTY));
> ```
> 
> to
> 
> ```
> Task.Run(() => StartLinuxWatcher(InsertedCallback, RemovedCallback, includeTTY), _cancellationTokenSource.Token);
> ```
> 
> This would be much easier than changing the native C code in Linux.
> 
> This should work because `IUsbEventWatcher` has:
> 
> ```
>     public void Dispose()
>     {
>         _cancellationTokenSource.Cancel();
>         _cancellationTokenSource.Dispose();
>     }
> ```
> 
> The only problem is that `udev_unref(g_udev);` would not be called. If you need that, I can try to find a native C solution. But then I will also try to find a native C solution for macOS, because I want it to behave the same way.

I don't think another CancellationTokenSource will work because the native function MonitorDevices blocks on select(2) so there is no chance for it to wake up, returning to C# code and checking the token's cancellation state. Maybe it's plausible to find a way to close(2) the udev monitor fd to have select(2) interrupted, which requires the fd to be passed into MonitorDevices instead of created by itself :-)

Jinjinov, 2023-05-19:
I made a new commit with `void StopLinuxWatcher()` function.

Can you take a look?

When I tried it in my Ubuntu, the test program stopped sending events.

zheng-yu-yang, 2023-05-19:
Thanks to the quick action!

In my test no more repeatedly reported USB insertion events. That's great.

I have to point out, after reading the code change, the fix may fail subtlely: it depends on setting the global variable runLinuxWatcher to 0 to exit from the loop, but if StopLinuxWatcher is called while the loop blocks on select, the loop will not exit immediately until some unsolicited USB things happen. The programmer may think as long as Dispose is called, it's safe to create another instance of UsbEventWatcher, and if he really does so before any new USB activity, the new instance will set runLinuxWatcher to 1 again, and the old monitoring task which is supposed to be gone will loose the chance to exit.

Guess there still needs to be a way to 'wakeup' the task from pending on select.

Jinjinov, 2023-05-19:
I made a new commit with:

```
int pipefd[2];
pipe(pipefd);
fcntl(pipefd[0], F_SETFL, flags | O_NONBLOCK);
FD_SET(pipefd[0], &fds);
write(pipefd[1], buffer, sizeof(buffer));
read(pipefd[0], buffer, sizeof(buffer));
```

Can you check it?

zheng-yu-yang, 2023-05-19:
Looks like you missed `#include <fcntl.h>`. It works well after the inclusion was added.

BTW, do you want to save the tasks created in `UsbEventWatcher.Start` as member fields of `UsbEventWatcher` and wait for them to finish in `UsbEventWatcher.Dispose`, so that make absolutely sure after an instance is disposed, everything is clean?

Thanks again.

Jinjinov, 2023-05-19:
I added `#include <fcntl.h>` but the program still crashes in Ubuntu 22.04 in VirtualBox.

Yes, I will save the tasks and wait for them in `Dispose`, that is a good idea.

zheng-yu-yang, 2023-05-19:
I tested on Ubuntu 18.04, and the test program was compiled by .Net 8 preview 3, NativeAOT.

Jinjinov, 2023-05-19:
I added a new commit - Dispose now waits for tasks to finish.

I must still find out why the program crashes in Ubuntu 22.04 in VirtualBox.

I am using .Net 7, without NativeAOT - perhaps VirtualBox is the problem.

Jinjinov, 2023-05-31:
Fixed for Linux in https://github.com/Jinjinov/Usb.Events/releases/tag/v10.1.1.0

I still have to fix it for macOS

Jinjinov, 2023-06-04:
Fixed for macOS in https://github.com/Jinjinov/Usb.Events/releases/tag/v10.1.1.1

closed 2023-06-04 by Jinjinov