Jinjinov/Usb.Events #29: Support for Raspberry Pi 4

tcochunk, 2023-07-03:
Thank you for the library....i was using it on Windows and it is very nice....easy to use and works as expected. Much appreciated.

I am wanting to use it on a Raspberry Pi, compiling the test project with .Net 6 (I see that the project is compatible with .Net Core 2 and higher) but when i execute the test program, i find no usb devices.  I have a mouse, keyboard, and a Raspberry Pico plugged in.  Here is my test project.....

`using IUsbEventWatcher usbEventWatcher = new UsbEventWatcher(startImmediately: true, addAlreadyPresentDevicesToList: true, usePnPEntity: true, includeTTY: true);

Console.WriteLine("Iterating over existing usb devices: " + usbEventWatcher.UsbDeviceList.Count());

foreach (var device in usbEventWatcher.UsbDeviceList)
{
    Console.WriteLine($"{DateTime.Now} Existing:" + Environment.NewLine + device + Environment.NewLine);
}
Console.WriteLine("Done Iterating over existing usb devices: " + usbEventWatcher.UsbDeviceList.Count());
`
I cant figure out what i am missing....any ideas?

Thanks

Jinjinov, 2023-07-04:
What OS are you using?

tcochunk, 2023-07-06:
Sorry for the delay in response...

I am running Raspbian Linux 11 bullseye the 32bit version\

Jinjinov, 2023-07-06:
Because `Usb.Events` uses native C for Linux and not C# it can behave differently than on Windows.

It is also possible that the native C code doesn't work on Raspbian Linux - I tested it only on Ubuntu.

Can you try testing with this code:

```
using Usb.Events;

 class Program
 {
     static void Main(string[] _)
     {
         using IUsbEventWatcher usbEventWatcher = new UsbEventWatcher();

         usbEventWatcher.UsbDeviceRemoved += (_, device) => Console.WriteLine("Removed:" + Environment.NewLine + device + Environment.NewLine);

         usbEventWatcher.UsbDeviceAdded += (_, device) => Console.WriteLine("Added:" + Environment.NewLine + device + Environment.NewLine);

         usbEventWatcher.UsbDriveEjected += (_, path) => Console.WriteLine("Ejected:" + Environment.NewLine + path + Environment.NewLine);

         usbEventWatcher.UsbDriveMounted += (_, path) =>
         {
             Console.WriteLine("Mounted:" + Environment.NewLine + path + Environment.NewLine);

             foreach (string entry in Directory.GetFileSystemEntries(path))
                 Console.WriteLine(entry);

             Console.WriteLine();
         };

         Console.ReadLine();
     }
 }
```

You can also try:
- on Ubuntu on a PC
- on Ubuntu on the Raspberry Pi
- on Windows IoT on Raspberry Pi

Try it with all USB devices you have - all USB flash drives, external hard drives, any keyboard, mouse, webcam, printer, joypad, ...
Try it on all USB ports

tcochunk, 2023-07-08:
So i tried keyboard, mouse, pico, and usb drive.....nothing triggered an event.  I believe Raspbian is based on the debian linux os but optimized for raspberry pi.

besides switching operating systems do you have any other ideas?

i know its asking ALOT but any chance you can test it on debian?

Jinjinov, 2023-07-08:
A few other ideas - make sure you have gcc `sudo apt install build-essential` and ludev `sudo apt-get install libudev-dev`:
- try running [Usb.Events.Test](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events.Test/Usb.Events.Test.csproj) directly from the source code, so there is no NuGet package - this way, the native Linux C code will compile on Raspberry Pi, see [here](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L57).
- you can also try running the [native C code](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Linux/main.c) by itself, without the C#, use this [makefile](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Linux/Makefile). This way you can put breakpoints directly into the native C code and see exactly what happens.

I will try to test it in Debian or, even better, on Raspbian for PC:
https://www.raspberrypi.com/software/raspberry-pi-desktop
https://projects.raspberrypi.org/en/projects/install-raspberry-pi-desktop

I hope that the reason is not that the native C code works on Intel, but not on ARM.

Jinjinov, 2023-07-09:
I found the reason: `Usb.Events` works only on `x64`, see [here](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L37)

I somehow forgot that, because I started the project a long time ago: April 19, 2020

Can you try with this: https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/ ?

tcochunk, 2023-07-18:
Thank you for the update.  I migrated my pi to 64bit and could NOT get it to work.  no usb devices are recognized.  I went back my 32 bit version of the OS and that also did not work.  I was using version 11.0.0.0

I compiled for AnyCPU and .net 7.  I published it for linux-arm (and linux-arm64 when i was using the 64 bit OS)

I was running the code you suggested I run.

any more thoughts?

Jinjinov, 2023-07-18:
That is my fault.

At first I compiled the native C code with:
```
gcc UsbEventWatcher.Linux.c -o UsbEventWatcher.Linux.so
```
Now I added:
```
gcc -m32 UsbEventWatcher.Linux.c -o UsbEventWatcher.Linux.so
gcc -m64 UsbEventWatcher.Linux.c -o UsbEventWatcher.Linux.so
```
And I thought that solved the problem.

Of course, to solve the problem for ARM, I will have to add:
```
gcc -march=armv7-a UsbEventWatcher.Linux.c -o UsbEventWatcher.Linux.so
gcc -march=armv8-a UsbEventWatcher.Linux.c -o UsbEventWatcher.Linux.so
```
I will do this as soon as I find the time to do it, perhaps today...

If you want to try it yourself, just add the gcc commands here: https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L54

Jinjinov, 2023-07-18:
I have somewhat bad news - installing ARM versions of `gcc`:
```
sudo apt-get install gcc-arm-linux-gnueabi
sudo apt-get install gcc-aarch64-linux-gnu
```
clashes with the package that enables to compile for 32 bit Intel on 64 bit Intel
```
sudo apt-get install gcc-multilib
```
That means the csproj file can not simultaneously include instructions to compile for ARM and Intel, which would be ideal.
To make it work, there would have to be 2 csproj files and 2 Linux virtual machines.

Even worse news:
```
sudo apt-get install libudev-dev:armhf libudev-dev:arm64
```
returns:
```
Unable to locate package libudev-dev:armhf
Unable to locate package libudev-dev:arm64
```
It has something to do with the fact that ARM packages are not available on Intel platforms by default.
I couldn't find a solution in the last 6 hours.
I don't know why I can install ARM versions of `gcc` on an Intel PC, but not the necessary libraries.

Right now I can think of only 3 solutions:
- I buy a Raspberry Pi
- You compile the source code on Raspberry Pi
- We find an expert who knows how to compile for ARM on an Intel PC

Jinjinov, 2023-07-19:
ChatGPT hallucinations led me down the wrong path, wasting A LOT my time.

I will try to find a solution for this.

Jinjinov, 2023-07-21:
This was the hardest issue so far, but I learned a lot! :)

I hope it works, I have no way to test it.

https://github.com/Jinjinov/Usb.Events/releases/tag/v11.0.0.1

tcochunk, 2023-07-29:
Sorry for the delayed response....work has been crazy.  I missed the ChatGPT comment until today....I chuckled and simultaneously felt bad for you and appreciative of your efforts.

You mentioned that what you said in the previous comment was not correct - i am not sure what part was not correct and I dont understand how using a Container (docker) could resolved the issue.

When i tested 11.0.0.1, i was not able to get it to work, I tried on 32 and 64 bit Pi OS.

Is there something I can do, you had mentioned compiling on the Pi (maybe that was the "wrong" part of your comment??)
Since i have the Pi, maybe i can do something to contribute to your project?  I would offer to send you a Pi, but they are so hard to get right now....its crazy.

Jinjinov, 2023-07-29:
There was a flood of my comments in this thread, I deleted most of them and truncated others...

It still doesn't work, huh... there is not much more I can do, but perhaps you can help - instructions at the end, you can skip there if details don't interest you.

To explain all the changes since the start of this issue - most of them are in the [Usb.Events.csproj](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj)

The problem: we need to compile `UsbEventWatcher.Linux.c` to `UsbEventWatcher.Linux.so` for platforms other than 64 bit Intel.

1. I ran `sudo apt-get install libudev-dev:i386` and `sudo apt-get install gcc-multilib` to my Ubuntu so that this line can execute:
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L59
it compiles a 32 bit `UsbEventWatcher.Linux.so` on a 64 bit Linux - but that only got us `UsbEventWatcher.Linux.so` for 32 bit Intel, not ARM.

2. because I couldn't get a gcc cross compiler for ARM to work (it would work if `UsbEventWatcher.Linux.c` wouldn't depend on `libudev` but it does, and I couldn't find a way to install an ARM version of `libudev-dev` on my Intel like I managed with `libudev-dev:i386` - but that was only for 32 bit Intel on 64 bit Intel, it works) - so I was left with the option to compile the ARM version in ARM environment - that is where Docker comes into play. Docker uses QEMU to emulate ARM on Intel, meaning you can have an ARM Ubuntu inside a Docker container running on Intel. In that ARM Ubuntu you can install gcc normally and add an ARM version of `libudev-dev` and use gcc to compile `UsbEventWatcher.Linux.c` to `UsbEventWatcher.Linux.so` for ARM. This happens here:
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L70
using these files:
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Dockerfile.arm
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Dockerfile.arm64
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/entrypoint.sh

The new NuGet has all these versions of `UsbEventWatcher.Linux.so` as you can see here if you open the `runtimes` folder:
https://nuget.info/packages/Usb.Events/11.0.1
you will see:
- linux-arm
- linux-arm64
- linux-x64
- linux-x86

I tested only `linux-x64` and `linux-x86` and they both work. But I can't test inserting physical USB devices to Docker - at least I don't know how to.

# TL;DR:

What you could do is relatively simple, but it would help a lot:

1. compile only the C code directly on ARM:
    - run `sudo apt-get install build-essential` on ARM
    - run `sudo apt-get install libudev-dev` on ARM
    - open [the Linux folder](https://github.com/Jinjinov/Usb.Events/tree/master/Usb.Events/Linux) in VS Code on ARM
    - debug [main.c](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Linux/main.c) in VS Code
    this will tell us if the C code even works on ARM.
2. if the C code works on ARM, you can test the C# on ARM
    - install C# Dev Kit extension in VS Code on ARM (it includes the C# extension)
    - install [.NET 7 SDK](https://learn.microsoft.com/en-us/dotnet/core/install/linux-ubuntu) on ARM
    - copy the `UsbEventWatcher.Linux.so` you got in the first step to `Usb.Events\Usb.Events\bin\Debug\netstandard2.0` and `Usb.Events\Usb.Events.Test\bin\Debug\net7.0` because the copy commands that do that in `Usb.Events.csproj` on Intel will not work on ARM
    - debug `Usb.Events.Test`

tcochunk, 2023-07-30:
I got the c library compiled on raspberry pi 64 bit using this link:

[https://stackoverflow.com/questions/14884126/build-so-file-from-c-file-using-gcc-command-line](https://stackoverflow.com/questions/14884126/build-so-file-from-c-file-using-gcc-command-line)

When I ran main.c, it seemed to be working as expected (see screen shot)
[image]

However, when i tried to build the C# code i got errors about gcc and -m32:

[image]

I am not sure how to resolve.....this seems to be coming from the project file from line 59 with Command="gcc -m32..."

it almost like it is tryingto build for x86 even though i am on an ARM???

any thoughts?

Jinjinov, 2023-07-31:
I am glad that you were able to run main.c and that it works! :)

You don't have to build from the terminal, the file that runs the gcc is the Makefile:
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Linux/Makefile
and it is already integrated into VS Code with these:
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Linux/.vscode/launch.json
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Linux/.vscode/tasks.json
so you can just select it from the top menu in VS Code: Terminal / Run build task
If you don't have that option in your VS Code, install the C/C++ extension and the Makefile Tools extension.

The problem with `-m32` is my mistake, I added tasks to Usb.Events.csproj that run the gcc to compile the C code before building the C# code. I didn't foresee that anyone would run the Usb.Events.csproj on ARM and I didn't think about it when I wrote you the instructions.

I changed the [Usb.Events.csproj](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj) to check for architecture before running the gcc, so it should work now.

tcochunk, 2023-08-05:
I tried a couple different things but could not get it to work.  This is what I did but i feel like I am screwing something up.

1. Built USBEventWatcher.Linux.o using make but added the fPIC CFLAG so i could convert it to a .so
2. Confirmed the c lib recognized the inserting and removing of Pico
3. Converted obj/USBEventWatcher.Linux.o to obj/USBEventWatcher.Linux.so because it was not created by default when building the c lib.
4. copied the .so file to Usb.Events/bin/Debug/netstandard2.0 (there was not a net7.0 folder)
5. built Usb.Events
6. built Usb.Events.Tests but got an error about undefined symbol udev_new

Like i said, i feel like i am screwing something up but dont know where to go.  Much of this work is not what i am used to (Linux, c, VSCode for C#, etc...)

[image]

Jinjinov, 2023-08-05:
Thank you for trying!

I know there is a lot involved, especially for someone who is not used to all technologies involved :)

If you carefully follow all these instructions, it should work:

Install these:
https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools
https://marketplace.visualstudio.com/items?itemName=ms-vscode.makefile-tools
https://marketplace.visualstudio.com/items?itemName=ms-dotnettools.csharp
https://marketplace.visualstudio.com/items?itemName=ms-dotnettools.csdevkit

Don't run `gcc` from the command line because you probably didn't use the right parameters, use `Terminal / Run Build Task...` to build:

[image]

or at least use the [Makefile](https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Linux/Makefile) where all the parameters are already defined.
To run `gcc` with a Makefile, simply type `make` in the command line in the same folder.

Copy the `UsbEventWatcher.Linux.so` to:
- `Usb.Events\Usb.Events\bin\Debug\netstandard2.0` and
- `Usb.Events\Usb.Events.Test\bin\Debug\net7.0`

Be careful, one is `Usb.Events\Usb.Events` and the other is `Usb.Events\Usb.Events.Test`

tcochunk, 2023-08-06:
I confirmed i had the 4 correct vscode tools. I opened the linux folder, selected the makefile, clicked Terminal->Run Build Task.

I was prompted with a tool selection -> Build or Build Debug.  I selected Build Debug but only got a UsbEventWatcher.Linux.o file was created in the Usb.Events/Usb.Events/Linux/obj directory.  

Is there something i should be doing to get the .so file or looking in a different place?

Jinjinov, 2023-08-06:
That is my fault again. I forgot that my script works only on Intel, not on ARM. There are so many different places in the build process where I never considered ARM. I'm on vacation this week, I will take a look when I get back home.

Jinjinov, 2023-08-06:
That is my fault again. I forgot that my script works only on Intel, not on ARM. There are so many different places in the build process where I never considered ARM. I'm on vacation this week, I will take a look when I get back home.

The easiest way you can make sure that you used the right parameters would be to look inside the entrypoint.sh (that is the script that ARM Ubuntu uses in Docker, so it should work on ARM)

Jinjinov, 2023-08-14:
Did you use the `-ludev` flag beside the `-fPIC` and `-shared` flags?

If you don't use `gcc -shared -ludev -fPIC` then you get `undefined symbol: udev_new`

closed 2023-08-30 by Jinjinov

tcochunk, 2023-09-30:
sorry to keep coming back.....i only get time intermittently to work on this...i found an issue in the csproj for compiling against arm64....around line 121 it says

exec condition isOsPlatform = linux and IsArm = true

but it is missing the longbit check.....because on 64 bit it tries to use -march=armv7-a+fp which i think is incorrect.....maybe i am wrong but when i put hte longbit check in, the buld correctly selected the msbuild cmd on line 129

i guess the commands could also just be reordered but that seems hacky

Jinjinov, 2023-10-01:
Could you use the latest version? I think that the line numbers don't match.

This line compiles to `./arm/` folder for 32 ARM on 32 and 64 bit systems
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L116
so no check is needed because a 64 bit system can compile for 32 bit

This line compiles to `./arm64/` folder for 64 ARM on 64 bit systems
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L124
so we need the 64 bit check because a 32 bit system can't compile for 64 bit

Did you try what I suggested the last time?

> Did you use the `-ludev` flag beside the `-fPIC` and `-shared` flags?

> If you don't use `gcc -shared -ludev -fPIC` then you get `undefined symbol: udev_new`

I think you were really close to making it work, because I can see `undefined symbol: udev_new` in your screenshot.

tcochunk, 2023-10-01:
you are correct the line numbers dont match, earlier i had added a couple log statements that messed up the line numbers. however we are talking about the same line numbers and the command on line 118 was giving me the error.  when i added the LongBit check the build correctly choose the path of your second reference and successfully built with line 126.

I then built the Test project and everything seems to be working as expected. HOORAY!!!

I will now test it with my code.  THANK YOU for your guidance and help!!!!

Jinjinov, 2023-10-02:
I am glad that you made it work! :)

Can you please tell me about your exact setup you are using now?
Which OS exactly?
Is your system now 32 or 64 bit?
What is the exact error you got on line 118?

If there are any mistakes in the `csproj` file, I want to fix them.

When using Intel, `gcc` can compile for 32 bit on a 64 bit system, that is why there is no bit check this line 
https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events/Usb.Events.csproj#L98

I was using the same logic for ARM, but I can't test it because I don't have an ARM.
If 64 bit ARM can't compile for 32 bit ARM then I will add the bit check.

tcochunk, 2023-10-16:
**Which OS exactly?**
using cat /etc/os-version
OS Debian GNU Linux 11

using uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023 aarch64 GNU/Linux

**Is your system now 32 or 64 bit?**
64 bit

**What is the exact error you got on line 118?**
I had some comments in so i think in the image below it is line 123 that was erroring
[image]

**Can you try adding -m32 to line 118 without the bit check in line 116?**
I had some comments in so i think in the image below it is line 121 that was erroring with this message
[image]

I made the following changes to the csproj file:
[image]

I made the following changes to the makefile:
[image]

In addition, when building the linux library on the PI, the so file was not create by the build process so i had to run a command to convert the .o to .so. This is what i ran in the linux obj directory:
gcc UsbEventWatcher.Linux.o -shared -fPIC -ludev -o UsbEventWatcher.Linux.so

I copied this file to the required locations per your instrunctions and your test project worked as expected.  i plugged in my pico and the test project found it, i unplugged the pico and the test project recognized the pico was unplugged.  However, when i tried to use that .so file in my test project, i could not get anything to work???  Any ideas? i was wondering if compiling the linux code in debug or release would matter???