Jinjinov/Usb.Events #30: Feature Request: Provide a way to toggle debug logging without relying on the DEBUG preprocessor directive

kmgallahan, 2023-07-17:
**Is your feature request related to a problem? Please describe.**

A large amount of text is always written to the debug output if the DEBUG preprocessor directive is defined. That is to say, if you are debugging a program then you will always see this text, regardless of if it is of any interest.

**Describe the solution you'd like**

Perhaps a settable bool with a default value of true could be checked on entering the logging methods.

**Describe alternatives you've considered**

No way to set a preprocessor directive by namespace. Can only be changed per file by modifying the source.

Jinjinov, 2023-07-18:
Yes, there is a lot of text in the debug output.

All of the text comes from here:
```
        private static void DebugOutput(ManagementBaseObject managementBaseObject)
        {
#if DEBUG
            System.Diagnostics.Debug.WriteLine(string.Empty);

            foreach (PropertyData property in managementBaseObject.Properties)
            {
                System.Diagnostics.Debug.WriteLine(property.Name + " = " + property.Value);
            }

            System.Diagnostics.Debug.WriteLine(string.Empty);
#endif
        }
```

I agree that a bool could be added to this method, but then the value of this bool would have to be changed if you wanted turn this output on and off - you would have to change one line of code.

You can solve this problem with one line of code, just add `return;` to the method:
```
        private static void DebugOutput(ManagementBaseObject managementBaseObject)
        {
            return;
#if DEBUG
            System.Diagnostics.Debug.WriteLine(string.Empty);

            foreach (PropertyData property in managementBaseObject.Properties)
            {
                System.Diagnostics.Debug.WriteLine(property.Name + " = " + property.Value);
            }

            System.Diagnostics.Debug.WriteLine(string.Empty);
#endif
        }
```

Or do you want to achieve something else?

kmgallahan, 2023-07-18:
I was just looking for a simple way to silence the output without forking the repo.

Jinjinov, 2023-07-18:
Oh, you mean without forking the repo. Yes, of course, I will find a solution for that.

Jinjinov, 2023-07-18:
For now, I just added `public static bool DisableDebugOutput { get; set; }` to `UsbEventWatcher`.

I will let you know when I release a new version, it should be soon.

I might implement `Microsoft.Extensions.Logging` in the future.

Jinjinov, 2023-07-18:
That is true :)

`public static bool EnableDebugOutput { get; set; }`

Jinjinov, 2023-07-21:
Solved in https://github.com/Jinjinov/Usb.Events/releases/tag/v11.0.0.1

closed 2023-07-21 by Jinjinov