Jinjinov/Usb.Events #33: Include osx-arm64 target runtime

slater1, 2023-11-16:
Thanks for a handy library.

On an M1 mac I get behavior like https://github.com/Jinjinov/Usb.Events/issues/32. I'm referencing the NuGet package. I get no notifications, no devices listed.

I noticed 11.0.1 does not include the `osx-arm64` runtime. Could that be why?

```
PS C:\Users\doug\.nuget\packages\usb.events\11.0.1\runtimes> ls

    Directory: C:\Users\doug\.nuget\packages\usb.events\11.0.1\runtimes

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----          11/15/2023    21:17                linux-arm
d----          11/15/2023    21:17                linux-arm64
d----          11/15/2023    21:17                linux-x64
d----          11/15/2023    21:17                linux-x86
d----          11/15/2023    21:17                osx-x64
```

slater1, 2023-11-16:
I compiled UsbEventWatcher.Mac.dylib by hand according to your instructions, placed it in my `runtimes/osx-arm64/native` folder and started receiving events.

I can manage for now, but pretty please can you add it to the package? Also happy to contribute the change.

Jinjinov, 2023-11-17:
Thank you for noticing this! I will take a look as soon as possible.

Jinjinov, 2023-11-17:
This was a huge oversight on my part and I am very thankful that you noticed this!

After I bought M1 mac I didn't modify the MSBuild script in `Usb.Events.csproj` to include `osx-arm64` into the NuGet.

[image]

It should work now:

[image]

Fixed in: https://github.com/Jinjinov/Usb.Events/releases/tag/v11.0.1.1

slater1, 2023-11-17:
Thanks @Jinjinov ! Super fast, I didn't even have time hack around the issue.

Sometimes it's fun to know who is using your lib. I'm this one in [FitEdit](https://www.fitedit.io) to detect when users plug in e.g. their Garmin devices.

closed 2023-11-17 by slater1

Jinjinov, 2023-11-17:
That is an interesting app you have there.

Respect for making it work on so many platforms - I know from personal experience ( https://ididit.today ) that it is not easy.

Good luck with your venture!

slater1, 2023-11-17:
I tried you app earlier - very neat. I was puzzled how you got it MAUI to work on Browser and Linux - and I wasn't sure exactly what your stack is. Blazor on top of Electron/Maui/Photino/Wpf maybe?

I went with Avalonia. It's pretty good. WASM has a long way to go though - wish I could serve up the app in a browser. I may take inspiration from you

Jinjinov, 2023-11-17:
Exactly that, I tried everything to see what works better: Blazor on top of Electron/Maui/Photino/Wpf

I started with Blazor WASM and went from there. I abstracted 99% of the code in a shared library. Electron is bloated and slow, so is Chromely. Maui Blazor hybrid is the only one that works in iOS and macOS. Photino is the only one that works in Linux. On Windows, Blazor works with WebView2 in WinForms and WPF, but WinForms crashes on exit. WPF compiles way faster than Maui, so I use it in debug.

I am very excited about Blazor United in .NET 8 that just came out this Tuesday. The ability to mix Blazor server and Blazor WASM in the same app is so amazing that I am rewriting the whole project and switching my hosting provider from Linux based one that provides only PHP, Python, Node.js hosting to another provider that has ASP NET Core hosting :)