Jinjinov/Usb.Events #36: Add sd-device backend for Linux

sonicbhoc, 2024-02-09:
While libudev is not currently in danger of being abandoned yet as far as I can tell, the documentation suggests that it has been superseded by sd-device. Perhaps adding an sd-device backend alongside the libudev one should be investigated.

Jinjinov, 2024-02-09:
Thank you for letting me know!

This would be a large task that would require a lot of time.

I am currently spending a lot of time on another open source project.

I will pin this issue and add the "help wanted" label.

Please let me know if the situation becomes more urgent and I will take the time to add sd-device.

sonicbhoc, 2024-04-30:
I'm making some headway on this. I'll have a PR up soon hopefully.

sonicbhoc, 2024-05-03:
Since both libudev and sd-device are both provided by systemd, and the systemd project itself has deprecated libudev, it makes sense to replace the libudev implementation with the sd-device implementation entirely, which will also reduce the amount of code to maintain.

Expect a pull request by the end of the day. Thanks for your help!

sonicbhoc, 2024-05-03:
@Jinjinov Do you mind if I clean up and refactor some of the other code in this lib while I'm looking at it? And if I can, am I allowed to give a major version bump and make breaking changes? I'd love to make some changes to adhere to the C# coding guidelines a bit more and make some things a bit more generic.

Jinjinov, 2024-05-04:
Hey @sonicbhoc !

I am really grateful for this large contribution and I will try to include all your changes into the project, but I would really appreciate it if you could make several smaller pull requests. The main reason is that I have a lot going on in my life right now - besides working on 2 other open source projects I am also buying a new apartment and I have some health problems that need my attention.

Thank you so much for doing all this, I would love to review it right away, but I am afraid it will take some time.

sonicbhoc, 2024-05-04:
The pull request is separated into multiple modest-size commits for the most part; I can't really simplify that one any more than I already have.

I wouldn't have included a massive effort like the one I suggested in the above comment, as it is unrelated. I just wanted permission to do so.

I'll make some issues with my suggestions and start implementing them. Good luck with your moving and get well soon.

Jinjinov, 2024-05-04:
Thank you so much!