Jinjinov/Usb.Events #45: Cannot use Usb.Events on MAC (arm64) in an osx-x64 targeted application with the latest release

bow-meow, 2024-05-22:
### Library version
11.1.0.1
### OS, version, architecture (32 bit / 64 bit)
MAC, Sonoma, arm64/x64
### Describe the bug
using an osx-x64 application on an m2 Mac (arm64) with version 11.1.0.1 of Usb.Events does not work.

### To Reproduce
1. Create a C# window application that targets osx-x64
2. Reference Usb.Events version 11.1.0.1
3. Create an instance of UsbEventWatcher and then dispose it.
4. Run the application and you will be met with a linker problem, the UsbEventWatcher.Mac.dylib needs to start with lib (libUsbEventWatcher.Mac.dylib) to be linked correctly
5. once that has been linked correctly and you run the application, upon disposing of usbeventwatcher, it will error saying it can't find libUsbEventWatcher.Mac.dylib x64 version, or that it can't use the current x64 version on an arm64 os.

### Expected behavior
Everything should just work.

### Additional context
Once you get passed the linking issue, version 10.1.0 of Usb.Events works fine, the latest does not.

Jinjinov, 2024-05-22:
Thank you for noticing this.

I have no free time these past few months and some health problems.

I am sorry to say that I don't know when I will look into this.

Jinjinov, 2024-09-20:
I have confirmed that this also happens on Apple Mac M1.

Jinjinov, 2024-09-20:
I have also confirmed that all version between version `10.1.0.0` which you confirmed as working and version `11.1.0.0` also work.

Version `11.1.0.1` is the only one that does not work.

Very little code was changed between `11.1.0.0` and `11.1.0.1` as most changes were for adding the documentation.

The only big change was

      <PackageReference Include="System.Management" Version="7.0.0" />

to

      <PackageReference Include="System.Management" Version="8.0.0" />

Jinjinov, 2024-09-20:
I tried `11.1.0.1` again and now it works without a problem.

I tried it with `ProjectReference` and with `PackageReference`, both work.

I don't know what changed, but I can't reproduce the problem anymore.

Jinjinov, 2024-09-21:
The only thing I can think of is that I deleted `bin` and `obj` folders when I tested each version.

That could have cleared some cached files.

Can you delete `bin` and `obj` folders and try again?

Jinjinov, 2025-01-24:
It is hard to manually compile the native code on macOS and Linux for x86, x64, arm32, arm64.

I will configure GitHub actions to compile all different native libraries and create another release.

Jinjinov, 2025-01-30:
It looks like I broke the code for Intel Mac when I started using ARM Mac M1.

Usb.Events on Intel Mac:
`10.1.1.1` - shows no info, throws exception on `Dispose`
`10.1.1.0` - shows no info, doesn't throw
`10.1.0.1` - shows info, doesn't throw

I don't know when I could fix this, I don't have an Intel Mac.

Jinjinov, 2025-02-08:
Fixed in https://github.com/Jinjinov/Usb.Events/releases/tag/v11.1.1.0

closed 2025-02-08 by Jinjinov