Jinjinov/Usb.Events #47: DllNotFoundException on MacOS x86_64

BlackCoyote, 2024-11-15:
### Library version
11.1.0.1
### OS, version, architecture (32 bit / 64 bit)
MacOS Sonoma X86_64
### Describe the bug
Calling UsbEventWatcher.Dispose() will throw the following DllNotFoundException;
```
Unhandled exception. System.DllNotFoundException: Unable to load shared library 'UsbEventWatcher.Mac.dylib' or one of its dependencies.In order to help diagnose loading problems, consider setting the DYLD_PRINT_LIBRARIES environment variable:
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64h' or 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib' (no such file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64h' or'x86_64'))
dlopen(/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib, 0x0001): tried: '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib' (no such file), '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib' (no such file)
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib' (no such file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib' (no such file)
dlopen(UsbEventWatcher.Mac.dylib, 0x0001): tried: 'UsbEventWatcher.Mac.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSUsbEventWatcher.Mac.dylib' (no such file), '/usr/lib/UsbEventWatcher.Mac.dylib' (no such file, not in dyld cache), 'UsbEventWatcher.Mac.dylib' (no such file), '/usr/local/lib/UsbEventWatcher.Mac.dylib' (no such file), '/usr/lib/UsbEventWatcher.Mac.dylib' (no such file, not indyld cache)
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib' (nosuch file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib' (no such file)
dlopen(/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib, 0x0001): tried: '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib' (no such file), '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib' (no such file)
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib' (no such file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib' (no such file)
dlopen(libUsbEventWatcher.Mac.dylib, 0x0001): tried: 'libUsbEventWatcher.Mac.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibUsbEventWatcher.Mac.dylib' (no such file), '/usr/lib/libUsbEventWatcher.Mac.dylib' (no such file, not in dyld cache), 'libUsbEventWatcher.Mac.dylib' (no such file), '/usr/local/lib/libUsbEventWatcher.Mac.dylib' (no such file), '/usr/lib/libUsbEventWatcher.Mac.dylib' (no such file, not in dyld cache)
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib.dylib' (no such file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/UsbEventWatcher.Mac.dylib.dylib' (no such file)
dlopen(/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/UsbEventWatcher.Mac.dylib.dylib' (no such file)
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib.dylib' (no such file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/UsbEventWatcher.Mac.dylib.dylib' (no such file)
dlopen(UsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: 'UsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSUsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/lib/UsbEventWatcher.Mac.dylib.dylib' (no such file, not in dyld cache), 'UsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/local/lib/UsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/lib/UsbEventWatcher.Mac.dylib.dylib' (no such file, not in dyld cache)
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/runtimes/osx-x64/native/libUsbEventWatcher.Mac.dylib.dylib' (no such file)
dlopen(/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/8.0.4/libUsbEventWatcher.Mac.dylib.dylib' (no such file)
dlopen(/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/Users/sandy/Projects/usbTest/usbTest/bin/Debug/net8.0/libUsbEventWatcher.Mac.dylib.dylib' (no such file)
dlopen(libUsbEventWatcher.Mac.dylib.dylib, 0x0001): tried: 'libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibUsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/lib/libUsbEventWatcher.Mac.dylib.dylib' (no such file, not in dyld cache), 'libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/local/lib/libUsbEventWatcher.Mac.dylib.dylib' (no such file), '/usr/lib/libUsbEventWatcher.Mac.dylib.dylib' (no such file, not in dyld cache)

   at Usb.Events.UsbEventWatcher.StopMacWatcher()
   at Usb.Events.UsbEventWatcher.Dispose()
[... cut ...]
usbEventWatcher.UsbDeviceRemoved += (, device) => Console.WriteLine("Removed");
usbEventWatcher.UsbDeviceAdded += (, device) => Console.WriteLine("Added");
Console.ReadKey();
Console.WriteLine("Closing app...");
usbEventWatcher.Dispose();
```

### Expected behavior
I expected the USB events to fire and the required DLL to be loaded.

### Additional context
I recognise that this issue seems similar to https://github.com/Jinjinov/Usb.Events/issues/45 but i was unable to find any solutions from the discussion in that issue.

Jinjinov, 2025-01-24:
Thank you for reporting this!

It is hard to manually compile the native code on macOS and Linux for x86, x64, arm32, arm64.

I will configure GitHub actions to compile all different native libraries and create another release.

Jinjinov, 2025-01-30:
It looks like I broke the code for Intel Mac when I started using ARM Mac M1.

Usb.Events on Intel Mac:
`10.1.1.1` - shows no info, throws exception on `Dispose`
`10.1.1.0` - shows no info, doesn't throw
`10.1.0.1` - shows info, doesn't throw

I don't know when I could fix this, I don't have an Intel Mac.

Jinjinov, 2025-02-08:
Fixed in https://github.com/Jinjinov/Usb.Events/releases/tag/v11.1.1.0

closed 2025-02-08 by Jinjinov