Jinjinov/Usb.Events #48: I want to get the port of the USB hot-swappable device

yyhLogin, 2024-12-03:
**Is your feature request related to a problem? Please describe.**
I want to get the port of the USB hot-plug device. Is there a unified API? Or does the author have a plan to implement this function?

**Describe the solution you'd like**
Not very familiar with Linux
**Describe alternatives you've considered**
I have used LibUsbDotNet, but its support for hot plugging is not very good

**Additional context**

Jinjinov, 2024-12-03:
Please describe in as much detail as possible what you want. What do you mean by "port"? Please give me an example.

yyhLogin, 2024-12-04:
[image]
On Windows, I just want to get the information on the graph. On Linux, I don't know how to describe it. It should be the location on the hub.

yyhLogin, 2024-12-05:
> It looks like this is only possible with C++ and not possible with C#
> 
> https://stackoverflow.com/questions/6416931/can-the-physical-usb-port-be-identified-programmatically-for-a-device-in-windows
> 
> > The [Win32_PnPSignedDriver](https://learn.microsoft.com/en-us/previous-versions/windows/desktop/legacy/aa394354(v=vs.85)) class has a "Location" field that seemed perfect (e.g. Port_#0001.Hub_#0010), but it only contains the location of the device when the driver was first loaded. Plugging the hardware into a different port does not update that field.
> 
> The problem: `Plugging the hardware into a different port does not update that field.`
> 
> C++ solution:
> 
> https://learn.microsoft.com/en-us/samples/microsoft/windows-driver-samples/usbview-sample-application/
> 
> https://github.com/microsoft/windows-driver-samples/tree/main/usb/usbview

Can you integrate this api in your project, as the implementation on Windows and Linux may be different