Jinjinov/Usb.Events #49: Crash on detaching a drive on Linux ARM64

M0ns1gn0r, 2025-01-24:
### Library version
11.1.0.1

### OS, version, architecture (32 bit / 64 bit)
Linux ARM64

### Describe the bug
After physically detaching of a USB Drive from our device, a segmentation fault exception was thrown in the native code, which led to termination of our dotnet process.

This doesn't reproduce easily (if at all) but since we've got a crashdump, maybe it will help finding the problematic code.

### Crash dump details from WinDbg

```
EXCEPTION_RECORD:  (.exr -1)
ExceptionAddress: 0000000000000000
   ExceptionCode: e0534947
  ExceptionFlags: 00000000
NumberParameters: 2
   Parameter[0]: 000000000000000b
   Parameter[1]: 0000000000000001

FAULTING_THREAD:  00000822

PROCESS_NAME:  dotnet

ERROR_CODE: (NTSTATUS) 0xe0534947 - <Unable to get error code text>

EXCEPTION_CODE_STR:  e0534947
EXCEPTION_PARAMETER1:  000000000000000b
EXCEPTION_PARAMETER2:  0000000000000001

ADDITIONAL_DEBUG_TEXT:  SOS.DLL is not loaded for managed code. Analysis might be incomplete

STACK_TEXT:  
0000ffff`7c55a840 0000ffff`be780cc4 libc_so+0xa0cc4
0000ffff`7c55a870 0000ffff`be4d2744 libcoreclr!PROCCreateCrashDump+0x268
0000ffff`7c55a8d0 0000ffff`be4d3978 libcoreclr!PROCCreateCrashDumpIfEnabled+0xab4
0000ffff`7c55a950 0000ffff`be4a69b4 libcoreclr!ZL22invoke_previous_actionP9sigactioniP9siginfo_tPvb+0x138
0000ffff`7c55a990 0000ffff`be4a5f74 libcoreclr!ZL15sigsegv_handleriP9siginfo_tPv+0x1b4
0000ffff`7c55ada0 0000ffff`bebab7ac linux_vdso_so!_kernel_rt_sigreturn+0x0
0000ffff`6d7dde10 0000ffff`be762ee4 libc_so+0x82ee4
0000ffff`6d7dde10 0000ffbe`a8d4191c UsbEventWatcher_Linux!GetChild+0x98
0000ffff`6d7dde70 0000ffbe`a8d424f0 UsbEventWatcher_Linux!GetLinuxMountPoint+0x94
0000ffff`6d7ddec0 0000ffff`88816a04 unknown!unknown+0x0

STACK_COMMAND:  .ecxr ; kb ; ** Pseudo Context ** Pseudo ** Value: 25 ** ; kb
SYMBOL_NAME:  libc_so+a0cc4
MODULE_NAME: libc_so
IMAGE_NAME:  libc.so.6
FAILURE_BUCKET_ID:  NOSOS_LINUX_SIGNAL_SIGSEGV_CODE_0x1_e0534947_libc.so.6!Unknown
OSPLATFORM_TYPE:  arm64
OSNAME:  Linux 0.0
FAILURE_ID_HASH:  {7d807a9a-aad9-5197-2118-3af6b9529fb3}
```

If I am not mistaken, the exception address 0x0000000000000000 indicates a null pointer dereference in native code, probably in UsbEventWatcher_Linux!GetChild.

Jinjinov, 2025-01-24:
Thank you for reporting this!

I will take a look at the native C source code and try to error proof it as much as possible.

Jinjinov, 2025-01-30:
Can you try with this file?

https://github.com/Jinjinov/Usb.Events/blob/master/Usb.Events.NuGet.Test/Usb.Events.11.1.1.nupkg

Just to make sure, before I publish it to NuGet.org

M0ns1gn0r, 2025-01-31:
Yes, I will test in in the beginning of the next week and write my observations.

M0ns1gn0r, 2025-02-03:
I tested the new package with our hardware and it looks like the crash has been fixed 💪 Previously I could reproduce the crash after about 20 attach/detach attempts, now it's still running even after 100 attempts.

Jinjinov, 2025-02-08:
Fixed in https://github.com/Jinjinov/Usb.Events/releases/tag/v11.1.1.0

closed 2025-02-08 by Jinjinov