Jinjinov/Usb.Events #52: crop release / AOT release failed

snow2zhou, 2025-05-13:
### Library version
11.1.1

### OS, version, architecture (32 bit / 64 bit)
 windows 11  64bit ，VS 2022 17.14.0 Priview 4.0

### Describe the bug

When I debug and run is  everything works fine, but when I use crop release, or when I use AOT release, the generated programme doesn't run properly and reports an error:

A type initializer threw an exception. To determine which type, inspect the InnerException's StackTrace property. 

If I don't use cropping release, it works fine.

How could I fix it?

### To Reproduce

private readonly UsbEventWatcher _watcher;

public Main()
{
try
{
    _watcher = new UsbEventWatcher(startImmediately: true);
    _watcher.UsbDeviceAdded += OnDeviceAdded;
}
catch (Exception ex)
{
    Console.Write(ex.Message);
}
}

private void OnDeviceAdded(object? sender, UsbDevice device)
{
    Console.Write("New Deiver!");
}

Jinjinov, 2025-05-13:
Issue arises because the Usb.Events library relies on static initializers and P/Invoke-based native hooks that get removed or broken when you trim or AOT compile your application.

P/Invoke is needed because part of Usb.Events is written with pure C code which is invoked from C#

In release builds with trimming or AOT enabled, the linker strips out types and methods that it deems unused, leading to a TypeInitializationException at runtime when the UsbEventWatcher cannot find its native dependencies.

To resolve this, you can disable trimming/AOT just for Usb.Events and keep it for the rest of your project.

A more fine grained solution would be to use attributes described in https://learn.microsoft.com/en-us/dotnet/core/deploying/trimming/prepare-libraries-for-trimming like:
- RequiresUnreferencedCode
- DynamicallyAccessedMembers
- DynamicDependency

But to do that, the exact reason for the TypeInitializationException must be known - which member exactly causes it.

snow2zhou, 2025-05-13:
My dear friends,
    Thank you for  your reply.
    It was happened around System.Management.

Jinjinov, 2025-05-13:
System.Management relies on built-in COM marshalling and reflection patterns that the .NET trimmer and Native AOT disable, so it is not compatible with either trimming or AOT.

Attempts to PublishTrimmed or PublishAot projects referencing System.Management will fail at runtime with InvalidProgramException or missing-method errors.

To resolve this, you can disable trimming/AOT just for Usb.Events and keep it for the rest of your project.