Jinjinov/Usb.Events #55: The application crashes with no exception when a USB flash drive is connected.

sgorozhankin, 2025-08-27:
### Library version
11.1.1
### OS, version, architecture (32 bit / 64 bit)
Mac mini M1
### Describe the bug
The application crashes with no exception when a USB flash drive is connected.

### To Reproduce
public static void Main()
    {
        Console.WriteLine("USB scanner (Usb.Events) — macOS\n");

        using IUsbEventWatcher usbEventWatcher = new UsbEventWatcher();
        usbEventWatcher.UsbDeviceAdded += OnAdded;
        usbEventWatcher.UsbDeviceRemoved += OnRemoved;

        foreach (var d in usbEventWatcher.UsbDeviceList)
        {
            DumpDevice(d);
        }

        using var exit = new ManualResetEvent(false);
        Console.CancelKeyPress += (_, e) =>
        {
            e.Cancel = true;
            exit.Set();
        };
        exit.WaitOne();
    }

    private static void OnAdded(object? sender, UsbDevice d)
    {
         Console.WriteLine("\n[+] Added device:");
         DumpDevice(d);
    }

    private static void OnRemoved(object? sender, UsbDevice d)
    {
        Console.WriteLine("\n[-] Removed device:");
        DumpDevice(d);
    }

    private static void DumpDevice(UsbDevice d)
    {
        Console.WriteLine($"  Name        : {d.DeviceName}");
        Console.WriteLine($"  Vendor      : {d.Vendor}");
        Console.WriteLine($"  Product     : {d.Product}");
        Console.WriteLine($"  Serial      : {d.SerialNumber}");
    }

Steps to reproduce the behavior:
1. Run the program
2. Connect USB Flash drive

### Exceptions (if any)
None. The app crashes without throwing an exception.

### Expected behavior
The app should not crash when a USB flash drive is connected.

### Additional context
The issue reproduces on multiple Macs. It occurs only when a USB flash drive is connected: if no flash drive is plugged in, everything works. I’m using a library to discover USB printers. The printers themselves work correctly.

Jinjinov, 2025-08-27:
Thank you for reporting this.

Unfortunately I don't have any free time right now because I am renovating my apartment.

Could you try debugging the source code?

hexxone, 2025-11-23:
https://github.com/Jinjinov/Usb.Events/pull/56

Maybe this is actually fixed with my PR :)

I also had some Issues on MacOs with verry long Device Path, due to USB Hubs...

Jinjinov, 2026-02-02:
@sgorozhankin can you try again with version https://github.com/Jinjinov/Usb.Events/releases/tag/v11.1.1.1

Pull request https://github.com/Jinjinov/Usb.Events/pull/56 fixed some issues on macOS

closed 2026-02-07 by Jinjinov